Medical-Robotics-Limbo/Electrical #2: Palm Hub

P7R7M, 2023-02-09:


P7R7M, 2023-02-15:
[image]

P7R7M, 2023-02-15:
[image]

kanand39, 2024-02-08:
02/05 - 02/09:
Tested current Palm Hub circuit and verified that the connections were right (validated that it was correct)
Tested circuit on a multimeter and oscilloscope while connected to power (laptop) and verified that the ADC is not functional
Reordered the SSOP16 PCB and found header pins for soldering next week

AnikethDandu, 2024-02-08:
.

kanand39, 2024-02-22:
02/12 - 0/2/21:
- Confirmed that ADC 108S052 is not functional
-  Set up Arduino to allow coding on RP 2040 (microcontroller being used)
- Checked schematic and found that circuit schematic was incorrect (CS was not connected to /CS, SCLK signal was incorrect, DIN, IN0, and DOUT were also incorrect)
       - Redesigned circuit schematic (ADC_kicad_V2 on repository)
- Looked through ADC test code 
- 02/20: Re-soldered ADC chip to SSOP16 
- 02/21: Created preliminary circuit with just ADC, inputted 4V (2.7V to 5.25V) from DC Power Supply (into Pin 2 -> analog VCC), GND from DC power supply to Pin 3 (GND on ADC), ran 5.6 MHz frequency into SCLK (Pin 16), and checked D_out (on oscilloscope)
             - Voltage shown on oscilloscope so ADC works