Medical-Robotics-Limbo/Electrical #3: EMG PCB

P7R7M, 2023-02-09:
The EMG sensor will basically be a custom version of this Myoware muscle sensor. We will be taking the raw EMG signal from the EMG electrodes, running it through a rectifier and a filter, then using an analog-to-digital converter for the microcontroller/RaspberryPi to use for the software team's algorithm.

[Myoware Guide](https://cdn.sparkfun.com/assets/a/3/a/f/a/AT-04-001.pdf)

[EMG processing using circuits](https://prostheticoss.wordpress.com/tag/emg/)

https://charleslabs.fr/en/project-OpenEMG+Arduino+Sensor

[Frequency Domain Intro](https://www.youtube.com/watch?v=fYtVHhk3xJ0)

[Frequency Domain Intro 2](https://www.youtube.com/watch?v=dBj89uSoqu4)

[Intro to types of filters](https://www.youtube.com/watch?v=rkwS6vigSyE&t)

[Cutoff frequency](https://www.youtube.com/watch?v=7T4U4dblCH8)

[Active op-amp high-pass and low-pass circuit](http://www.emagtech.com/wiki/index.php/Advanced_Tutorial_Lesson_2:_Designing_Active_Sallen-Key_Filters)

P7R7M, 2023-02-13:
Whoever is doing this needs to design an EMG board so that it can be printed on the voltera. It should be a two layer board, here are some building constraints https://docs.voltera.io/v-one/v-one-fundamentals/circuit-design-guidelines, and an example board is here https://charleslabs.fr/en/project-OpenEMG+Arduino+Sensor

sethwobig3, 2023-02-16:
EMG parts based on Charles Lab surface mount design:
https://docs.google.com/spreadsheets/d/1GXmlTlyWqPjLi16XgMQEayzVkyHjtmpGHZKpF-S9OZY/edit?usp=sharing

P7R7M, 2023-02-22:
Just adding a note that our eventual goal is to have one board reading multiple emg's at the same time

stran47, 2023-04-19:
circuit design of PCB board from 3110 quant lab (what we've been using for EMG so far); need to design with just the essentials (blue boxes)

[image]

[image]