Medical-Robotics-Limbo/Electrical #4: Power Management System

P7R7M, 2023-02-09:
https://learn.sparkfun.com/tutorials/power-delivery-board---usb-c-qwiic-hookup-guide/all

[image]

Useful links on USB-C Power Delivery (PD):
https://www.youtube.com/watch?v=OwAZqJ4wpJg

Buck vs LDO Regulator:
https://www.youtube.com/watch?v=giGRrODKJSE&t=139s
GOAL: feed 8.4V and get out 5V and 3.3V (for now)

https://docs.google.com/document/d/1gVOqEcoC50Bx6NVB0qYACGVpzWjTPaRM8ukTt0245ng/edit?usp=sharing

P7R7M, 2023-02-16:
Trying to start by modifying google sheets in google drive to calculate battery needed and length of life, mechanical says they need 6 motors and 3 servos

P7R7M, 2023-02-16:
rough space specs, def over, are 60mm diameter by 120 mm depth

P7R7M, 2023-02-22:
[image]

P7R7M, 2023-02-27:
biggest rectangle battery that can fit is 42x42x120 mm

P7R7M, 2023-03-02:
decided on this battery for now: https://www.amazon.com/3000mAh-Battery-WLTOYS-Upgrade-Charger/dp/B0887TLWWB/ref=asc_df_B0887TLWWB/?tag=hyprod-20&linkCode=df0&hvadid=475772725632&hvpos=&hvnetw=g&hvrand=11555683523190682289&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1015254&hvtargid=pla-1144675493427&psc=1, going to add to purchasing sheet

P7R7M, 2023-03-07:
exploring this as a buck converter option, gonna focus on the conversion side before the chargin side for priority of a working prototype: https://www.digikey.com/en/products/detail/texas-instruments/TPSM84203EAB/7561619?utm_adgroup=Texas%20Instruments&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Shopping_Supplier_Texas%20Instruments&utm_term=&utm_content=Texas%20Instruments&gclid=CjwKCAiAu5agBhBzEiwAdiR5tLjvhSbuddLsKxWdufY2yPBvYlPLwKL3zgokFA7bn1L80aDD2xRDVRoCH1QQAvD_BwE

P7R7M, 2023-03-08:
a SUPER cool battery: https://www.i-biomed.com/flexcell.html

spoma3, 2023-04-12:
[Buck Converter Stats.xlsx](https://github.com/Medical-Robotics-Limbo/Electrical/files/11216766/Buck.Converter.Stats.xlsx)

P7R7M, 2023-04-17:
We should take 5V linear regulators from the HIVE (REG_5V_LDL1117S50R) and use that to convert the LiPo voltage to 5V to power the microcontroller.