Medical-Robotics-Limbo/Electrical #6: Validate Slip Sensor Data

P7R7M, 2023-02-09:
The pictures below show the slip sensor schematics. Top (red) and bottom (blue).

[image]
[image]

The slip/force sensor collects values and sends them to a microcontroller, so it displays them in Arduino. We switched the microcontroller from an Arduino to raspberry pi Pico (RPP). The next goal is to map out the values from the sensor to make sense of them.

P7R7M, 2023-02-14:
find the sensor bruh

stran47, 2023-02-22:
current set up of slip sensor board (not working, not sure if it is a sensor, board, or code problem) - trying to troubleshoot (going to test on Arduino nano board)

[image]
[image]