Medical-Robotics-Limbo/Electrical #7: Validate Force Sensor Data

P7R7M, 2023-02-09:


P7R7M, 2023-03-08:
https://www.sparkfun.com/products/8685

WangIxihan, 2023-03-08:
03/08 - Using Datasheet from link above, constructed the circuit (without Vt) and got data on the serial monitor and plotter. 
I don't have an ipad to annotate. Op-Amp used is TL071CP
- Purple wire is from one + rail (3.3V)
- Yellow wire is from Vout in Op-Amp,6 to Vsys (39) on raspberry pi pico.
- Green wire connected from 3+ to ground
- Connected random resistor value Rf from 2- to 6out on opamp
- small red wire - Vcc+ 5.5V from 7 opamp
- orange wire from 4 to GND opamp.

WangIxihan, 2023-03-27:
Update the reference resistor to 10 kOhm. Can't find the force sensor arduino code in the github repository anymore.

WangIxihan, 2023-04-03:
Connected board to voltage supply and drive voltage - got something but it seems like the force sensor response time is not always immediate.