Medical-Robotics-Limbo/Electrical #8: Map Potentiometer Values for Motor Position Sensing

P7R7M, 2023-02-09:


P7R7M, 2023-02-13:
https://www.mouser.com/new/bourns/bourns-3382-sensors/?gclid=CjwKCAiA3KefBhByEiwAi2LDHLpJi657SA1dDwvo4-f2De_XS81kxyqe0tUnJilI7-ytcvRl_ApU_BoCkwoQAvD_BwE

P7R7M, 2023-02-13:
motor link: https://www.handsontec.com/dataspecs/motor_fan/GA12-N20.pdf

P7R7M, 2023-02-13:
potentiometer link: https://www.bourns.com/docs/Product-Datasheets/3382.pdf

P7R7M, 2023-03-13:
Look into either finding a different potentiometer, or buying motors with them built in: https://www.amazon.com/dp/B08DKJT2XF/ref=sspa_dk_detail_5?psc=1&pf_rd_p=88097cb9-5064-44ef-891b-abfacbc1c44b&pf_rd_r=P8T0G5Z2JJBEC1RA7ARV&pd_rd_wg=bfS4c&pd_rd_w=VijJg&content-id=amzn1.sym.88097cb9-5064-44ef-891b-abfacbc1c44b&pd_rd_r=0d7f9f4c-93a6-4a7f-9a0b-f7e008a9b313&s=hi&sp_csd=d2lkZ2V0TmFtZT1zcF9kZXRhaWw&smid=A1RXNZT4AK0KHN&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFFVzg0NFNWSldZQVomZW5jcnlwdGVkSWQ9QTA2NDYxMTJDV0M0N0hYVzkyVDAmZW5jcnlwdGVkQWRJZD1BMDkyNjM0ODFYUTZLQ00yVllUUUkmd2lkZ2V0TmFtZT1zcF9kZXRhaWwmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

jamesdeocampo, 2023-03-13:
Possible Potentiometers:
https://www.mouser.com/c/passive-components/potentiometers-trimmers-rheostats/potentiometers/#:~:text=A%20potentiometer%20is%20an%20adjustable,of%20current%20to%20the%20circuit.

Constraints:
40 mL to 44 mL, anything that can fit in it. The smaller, the better.
**Rotary potentiometer** -> transition from 330 to full 360 degrees of rotation through **Continuous Potentiometer**. old one is: GA-12DC motor
Find something with a smaller encoder stack
Test the power and torque of the other two motors. Are they worth it?
Try to get everything done by April

P7R7M, 2023-04-19:
wired up the motor that has an encoder on it, think i wired it correctly but cant read any data.