Medical-Robotics-Limbo/Electrical #24: Slip and Force Sensor

AnikethDandu, 2024-02-19:


WangIxihan, 2024-02-22:
Slip and Force Sensor

Bought
Hall Sensor
Light Sensor (to replace the old phallic-shaped sensors that don’t work)

Before then, I will order some cheap piezoelectric sensors to see if we can set up a circuit using either arduino and raspberry pi 

Complete this by the end of the semester.
Demonstrate that the piezoelectric sensor somehow works.

Transfer data into arduino IDE.
Write up a procedure 

11/27/23
Piezoelectric sensors are noisy at detecting blood pressure, so maybe that can be improved with fine tuning. But, I successfully removed the resonant frequency from the air.
Instrumentation amplifier
Band-pass filter
Any more circuit components?
Goal for Spring:
Shrink my design
Test using one finger.

Spring 2024 Goals:
Investigate the viability of higher-order Chebyshev filters
Analog to digital converter
Investigate the viability of incorporating Hall Sensors into our design
Shrink and test for one finger.

Analog-Digital Converter:
Converts voltage reading into digital reading to be processed in a microcontroller
10-bit IC ADC

https://www.electronics-tutorials.ws/electromagnetism/hall-effect.html
‘Biasing’ magnet for Hall effect

WangIxihan, 2024-02-22:
PDF manual:
[BioTac® Product Manual](https://www.syntouchinc.com/wp-content/uploads/2018/08/BioTac-Manual-V.21.pdf)
BioTac Website:
[Robotics & Prosthetics | SynTouch (syntouchinc.com)](https://syntouchinc.com/robotics/)
Patent:
[1499065777900190136-US20160025615A1 (storage.googleapis.com)](https://patentimages.storage.googleapis.com/b2/4c/a1/e9371e23a2cb25/US20160025615A1.pdf)

Sensors in Focus:
Impedance Sensing Electrodes
As forces are applied to the skin, the skin and fluid deform. Changes in impedance as the fluid deforms are detected by an array of electrodes on the surface of the BioTac core.
Hydro-Acoustic Pressure Transducer
As objects slide across the surface of the BioTac, they generate vibrations that are detected by the sensor inside the core

Impedance Sensing Electrodes
Basis: A voltage divider with 4 common excitation electrodes, sensing electrode, and a reference to a 10kΩ load resistor
A 3.3V pulse will be sent from the microcontroller through the excitation electrodes to the sensing electrode 

Questions: 
What are excitation electrodes?
Electrodes that receive electrical signals
What are the sensing electrodes that are being used and can we source them?
A normal sensor (force sensing resistor)
[Amazon.com : force sensor](https://www.amazon.com/force-sensor/s?k=force+sensor) 

Hydro-Acoustic Pressure Transducer: https://www.benthowave.com/
Website contains everything about hydrophone sensors and transducers
Hemispheric Transducer is possible, omnidirectional is a candidate
Frequency response for the Biotac sensor is 10 - 1040 Hz.

Hall Sensors: sensors that sense magnetic fields. Output voltage is directly proportional to the strength of the magnetic field.

I think we can start building. The pressure sensor seems to be a wheatstone bridge with one variable resistor.
Why the pressure sensor is a Wheatstone Bridge: it acts like a balancer, readjusting the variable resistor to accommodate for changes in the detecting resistor.

Circuit Schematic and Parameters:
P_DC op amp (fluid pressure): MPX2101A
P_AC op amp (micro vibrations): 

Mon, Oct 2:
Just 2 op-amps to start with.
Resistor load is 10k ohms.
Literally just electrodes → literally just points of contact for electricity.
Flexible Circuit Model pinout is in paragraph below the manual.